riscv-collab/riscv-openocd #1164: Why memory read speed is much slower than memory write speed？

lz-bro, 2024-11-13:
> \>dump_image test1.bin 0xffffffff80003e3e 1048576
>dumped 1048576 bytes in 42.666645s (24.000 KiB/s)
> \>load_image test1.bin 0xffffffff80003e3e
>downloaded 1048576 bytes in 10.778667s (95.002 KiB/s)
I conducted a memory block read and write test and found that the memory read speed is relatively slow. Why is that?

TommyMurphyTM1234, 2024-11-13:
Might be worth capturing and attaching (not copying and pasting) a verbose `-d3` log for the two scenarios for comparison?

lz-bro, 2024-11-13:
> Might be worth capturing and attaching (not copying and pasting) a verbose `-d3` log for the two scenarios for comparison?

The logs have been attached, thank you.
[load.zip](https://github.com/user-attachments/files/17728797/load.zip)
[dump.zip](https://github.com/user-attachments/files/17728798/dump.zip)