riscv-collab/riscv-openocd #1166: Multi-die RISC-V core configuration file

zhangdujiao, 2024-11-14:
For multi-die RISC-V core debugging, assuming each die integrates DTM (JTAG TAP) with the same TAPID, and they are connected as "daisy chain".
How to make the configuration files properly, so that the debugger can correctly access different dies and cores?

Could the "-chain-position" command be used for the same TAPID when creat newtap?

en-sc, 2024-11-14:
1. JTAG chain should be declared as usual. See [OpenOCD guidelines](https://openocd.org/doc-release/html/Config-File-Guidelines.html#Adding-TAPs-to-the-Scan-Chain) for reference.
2. Create the targets via `target create` command. `-chain-position` specifies to which TAP the target corresponded.
3. If there are multiple RISC-V Debug Modules under a single RISCV JTAG DTM, `-dbgbase` is used to specify the base DMI address of the Debug Module. If there is only one Debug Module on each DTM, `-dbgbase` can be omitted.

zhangdujiao, 2024-11-14:
If two dies with the same TAPID are in one daisy chain, IRLEN is 5, and the target die for debugging is die1, then each JTAG instruction shift operation would require sending 10 bits in total. The lower 5 bits would be the value for the die0 TAP, and the upper 5 bits would be for the die1 TAP. Is that correct?

In general, if there are N dies connected in series, each with an IRLEN of 5, the JTAG chain would require a shift of 5 × N bits for each access. This ensures that each die in the chain receives the correct segment of the instruction.

en-sc, 2024-11-14:
> ... The lower 5 bits would be the value for the die0 TAP, and the upper 5 bits would be for the die1 TAP. Is that correct?

The other way around. Generally, JTAG shifts LSB first. Therefore the lower bits will go to the furthest TAP. So if your chain is:
```
|    TAP 0    |    TAP 1   |
----------------------------
-> TDI | TDO -> TDI | TDO ->

```

Then the lower 5 bits will go to TAP 1.

> In general, if there are N dies connected in series, each with an IRLEN of 5, the JTAG chain would require a shift of 5 × N bits for each access.

To be exact, IR shift will only be required to change the IR. In general, this is rarely needed -- most of the time DMI instruction is selected (with BYPASS selected on other TAPs) and the access is done via DR shift.

closed 2024-11-15 by zhangdujiao