riscv-collab/riscv-openocd #1172: How to read memory to get some information before target create？

lz-bro, 2024-11-22:
For multi-cluster RISC-V core debugging, all clusters belong to the same tap. Actually, not all clusters may be used because there is partial-good situation. So how to read memory to get some information before `target create`.

TommyMurphyTM1234, 2024-11-22:
> For multi-cluster RISC-V core debugging, all clusters belong to the same tap.

What exactly do you mean by "cluster" here?

> Actually, not all clusters may be used because there is partial-good situation.

What does "partial-good situation" mean?

> So how to read memory to get partial-good information before `target create`.

As far as I know the OpenOCD memory access commands require that the target has been created before they can be used.

- https://openocd.org/doc/html/General-Commands.html#:~:text=15.3%20Memory%20access%20commands

Maybe you can use raw/low level JTAG commands to do what you need? 

- https://openocd.org/doc-release/html/JTAG-Commands.html

aap-sc, 2024-11-22:
@lz-bro

**general note:** this tracker is probably not the best place to ask these questions, since it looks like a general OpenOCD question, not RISC-V specific one. Probably if you ask in OpenOCD mailing list you can get a more detailed answers.

**back to your question:** if you mean that some targets require additional initialization before the can be accessed/examined, then OpenOCD supports "deferred examine". See "-defer-examine" argument to `target create` in OpenOCD documentation. Maybe this could work for you?

Additionally we already had questions related to "how to read memory before init". Here is an example of a discussion: https://github.com/riscv-collab/riscv-openocd/issues/1038 . If this is something that you want, you could take a look there (though, there are no detailed instructions there, only general guidance).

lz-bro, 2024-12-11:
@TommyMurphyTM1234 @aap-sc I can not use irscan/drscan commands to read memory before target create, because irscan/drscan commands must be used after init and target create command must be used before init.

lz-bro, 2024-12-11:
> @lz-bro you can. Just create taps and don't create targets

If I want to use irscan/drscan commands to read memory get some information, then target create based on the information， how can I do this?

aap-sc, 2024-12-11:
@lz-bro it depends on what you need to do. You have several options.

1. Write a script that just launches OpenOCD two times. The first time you do low-level initialization and write some information in  a file (jimtcl allows you to do that). Then run OpenOCD for the second time passing all the necessary parameters to your configs. This flow works quite well from my experience.
2. Alternatively , you can still create the targets but request a deferred examine (see `-defer-examine`) flag. Then you don't need to run OpenOCD several times and you can do everything you need in one session.

lz-bro, 2024-12-11:
@aap-sc Thanks for reply. For the second option,  it is possible that some targets are not needed in the end base on the information.
for example：
If I use irscan/drscan commands to read memory and get `0b0011`, I would target create 2 harts.
if I get `0b1111`, I would target create 4 harts.

aap-sc, 2024-12-11:
> @aap-sc Thanks for reply. For the second option, it is possible that some targets are not needed in the end base on the information. for example： If I use irscan/drscan commands to read memory and get `0b0011`, I would target create 2 harts. if I get `0b1111`, I would target create 4 harts. Can this be handled in one session?

No, I don't think so. AFAIK OpenOCD does not allow you to dynamically modify your target list. In your case I would recommend you to go with option 1.

aap-sc, 2024-12-11:
> No, I don't think so. AFAIK OpenOCD does not allow you to dynamically modify your target list. In your case I would recommend you to go with option 1.

Do note, however, that .tcl itself supports this kind of scripting. So in theory you could invoke openocd from the configuration file itself. So in theory it may be possible to write this in a manner transparently to the user

lz-bro, 2024-12-12:
> > > @aap-sc Thanks for reply. For the second option, it is possible that some targets are not needed in the end base on the information. for example： If I use irscan/drscan commands to read memory and get `0b0011`, I would target create 2 harts. if I get `0b1111`, I would target create 4 harts. Can this be handled in one session?
> > 
> > 
> > No, I don't think so. AFAIK OpenOCD does not allow you to dynamically modify your target list. In your case I would recommend you to go with option 1.
> 
> Do note, however, that .tcl itself supports this kind of scripting. So in theory you could invoke openocd from the configuration file itself. So in theory it may be possible to write this in a manner that is transparent to the user

How can I invoke openocd from the configuration file itself，Use `exec` command Error.

aap-sc, 2024-12-12:
```
How can I invoke openocd from the configuration file itself，Use exec command Error.
```

Could you provide more context? I don't understand the question.

lz-bro, 2024-12-13:
> Do note, however, that .tcl itself supports this kind of scripting. So in theory you could invoke openocd from the configuration file itself. So in theory it may be possible to write this in a manner that is transparent to the user

`Do note, however, that .tcl itself supports this kind of scripting. So in theory you could invoke openocd from the configuration file itself. So in theory it may be possible to write this in a manner that is transparent to the user`

Is it means  that I can invoke openocd from the .tcl, but I did not succeed, could you help me? like,
```

set command "$OPENOCD_PATH/openocd -f $CONFIG_FILE "

set output [exec $command]

set _CHIPNAME riscv
jtag newtap $_CHIPNAME cpu -irlen 5 -expected-id $TAPID

# handle output to determine how many targets to create
...

```

aap-sc, 2024-12-13:
> `Do note, however, that .tcl itself supports this kind of scripting. So in theory you could invoke openocd from the configuration file itself. So in theory it may be possible to write this in a [...]

here is an example that works for me (file `test.tcl`):

```
echo "hello, this is an OpenOCD!"

set CmdResult [exec build/Release/install_openocd/openocd/bin/openocd -c "echo \"     hello from second OpenOCD\"" -c shutdown]

echo "Below we have the help from additional openocd run:"
echo "$CmdResult"

echo "bye!"

shutdown
```

I run it as:

```
build/Release/install_openocd/openocd/bin/openocd -f test.tcl
```

this produces the following output :

```

Open On-Chip Debugger 0.12.0+dev-aap_sc_bleh_blue_experimental-g84705d15-dirty (2024-12-09-20:50)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
hello, this is an OpenOCD!
Below we have the help from additional openocd run:
Open On-Chip Debugger 0.12.0+dev-aap_sc_bleh_blue_experimental-g84705d15-dirty (2024-12-09-20:50)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
     hello from second OpenOCD
shutdown command invoked
bye!
shutdown command invoked
```

If you need something more complex -you can try to build on top of this example... You can also try to provide more context on the nature of the failure (I still don't quite get what do you mean by "did not succeed").

lz-bro, 2024-12-20:
@aap-sc 
here is an example that works for me (file test.tcl):
```
set OpenOCDArgs [list \
     -c "adapter driver jlink" \
     -c "transport select jtag" \
     -c "adapter speed  2000" \
     -c "jtag newtap $_CHIPNAME_0 cpu -irlen 5 -expected-id 0x10000E21" \
     -c "jtag newtap $_CHIPNAME_1 dap -irlen 4 -expected-id 0x4ba06477" \
     -c "jtag newtap $_CHIPNAME_3 cpu -irlen 5 -expected-id 0x10000E21" \
     -c "target create $_CHIPNAME_3.cpu riscv -chain-position $_CHIPNAME_3.cpu" \
     -c "$_CHIPNAME_3.cpu configure -event examine-end { halt; echo \[read_memory 0x53c0703e8 32 1\]; shutdown }"]

set Commands "C:/Users/fpgauser/Desktop/openocd/bin/openocd"
  foreach Arg $OpenOCDArgs {
    append Commands " "
    if {[regexp {\s+} $Arg dummy]} {
      append Commands "\"$Arg\""
    } else {
      append Commands "$Arg"
    }
  }
echo "$Commands"
set CmdResult [exec $Commands]
```
I run it as: `C:/Users/fpgauser/Desktop/openocd/bin/openocd -f test.tcl` , Error like that:
this produces the following output :
```
Open On-Chip Debugger 0.12.0+dev-gd2acb17 (2024-09-06-07:34)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
C:/Users/fpgauser/Desktop/openocd/bin/openocd -c "adapter driver jlink" -c "transport select jtag" -c "adapter speed  2000" -c "jtag newtap ACPU cpu -irlen 5 -expected-id 0x10000E21" -c "jtag newtap DAP dap -irlen 4 -expected-id 0x4ba06477" -c "jtag newtap RCPU cpu -irlen 5 -expected-id 0x10000E21" -c "target create RCPU.cpu riscv -chain-position RCPU.cpu" -c "RCPU.cpu configure -event examine-end { halt; echo [read_memory 0x53c0703e8 32 1]; shutdown }"
spacemit_k1m_chain_rcpu_test.cfg:53: Error: couldn't exec "C:/Users/fpgauser/Desktop/openocd/bin/openocd -c "adapter driver jlink" -c "transport select jtag" -c "adapter speed  2000" -c "jtag newtap ACPU cpu -irlen 5 -expected-id 0x10000E21" -c "jtag newtap DAP dap -irlen 4 -expected-id 0x4ba06477" -c "jtag newtap RCPU cpu -irlen 5 -expected-id 0x10000E21" -c "target create RCPU.cpu riscv -chain-position RCPU.cpu" -c "RCPU.cpu configure -event examine-end { halt; echo [read_memory 0x53c0703e8 32 1]; shutdown }""
in procedure 'script'
at file "embedded:startup.tcl", line 28
at file "spacemit_k1m_chain_rcpu_test.cfg", line 53
```

aap-sc, 2024-12-20:
@lz-bro 

```
here is an example that works for me (file test.tcl):
```
I assume that it does not.

This more like a question about .tcl... I suspect that you are using `exec` incorrectly. Try to change the last line to:

```
set CmdResult  [exec {*}$Commands]
```

TommyMurphyTM1234, 2024-12-20:
> I assume that it does **not** work. Right?

Yes, this is confusing...

> @aap-sc
> here is an example that works for me (file test.tcl):

...

> I run it as: C:/Users/fpgauser/Desktop/openocd/bin/openocd -f test.tcl , Error like that:
> this produces the following output :
>
> ```
> Open On-Chip Debugger 0.12.0+dev-gd2acb17 (2024-09-06-07:34) [...]

lz-bro, 2024-12-24:
> @lz-bro
> 
> ```
> here is an example that works for me (file test.tcl):
> ```
> 
> I assume that it does **not** work. Right?
> 
> This more like a question about .tcl... I suspect that you are using `exec` incorrectly. Try to change the last line to:
> 
> ```
> set CmdResult  [exec {*}$Commands]
> ```

It is OK, thank you very much!

closed 2025-01-23 by lz-bro