riscv-collab/riscv-openocd #1182: How to use trace action in RISC-V target triggers?

Alan-19950616, 2024-12-10:
In the current OpenOCD code, the trace related action is not used.
```
/*
 * trace on:
 */
#define CSR_MCONTROL_ACTION_TRACE_ON        2
/*
 * trace off:
 */
#define CSR_MCONTROL_ACTION_TRACE_OFF       3
```

JanMatCodasip, 2024-12-10:
Hi @Alan-19950616, 

In riscv-openocd, there are currently no commands that would set up triggers with trace actions for you.

Still, you can set up such a trigger manually:

1) Reserve the trigger to prevent OpenOCD from touching it - command [riscv reserve_trigger](https://github.com/riscv-collab/riscv-openocd/blob/ea8f9d51954b979ff6b4d90afa70352763199b63/src/target/riscv/riscv.c#L5552)
2) Use standard CSR register write commands to set up your trigger: `reg tselect <value>`, `reg tdata1 <value>`, ...

Alan-19950616, 2024-12-11:
@JanMatCodasip @aap-sc Thanks for your replies.

The method you propose does successfully configure TRACE_ON/TRACE_OFF, but OpenOCD does not know that these Triggers are being used as TRACE_ON/TRACE_OFF functions, and therefore will conflict with normal hardware breakpoints.

aap-sc, 2024-12-11:
@Alan-19950616 have you tried  `[riscv reserve_trigger](https://github.com/riscv-collab/riscv-openocd/blob/ea8f9d51954b979ff6b4d90afa70352763199b63/src/target/riscv/riscv.c#L5552) mentioned by  @JanMatCodasip ? That command is designed to address this issue.

Alan-19950616, 2024-12-16:
Thanks. I will close this.