riscv-collab/riscv-openocd #1185: error: 'PCI_EXT_CAP_ID_VNDR' undeclared

Alan-19950616, 2024-12-16:
I'm having problems trying to compile the latest OpenOCD here at de20c2ad5faa0bc411b86bd6db69186b8c2ac979.

```
Work/openocd-build/openocd.git/src/jtag/drivers/xlnx-pcie-xvc.c: In function 'xlnx_pcie_xvc_init':                         
Work/openocd-build/openocd.git/src/jtag/drivers/xlnx-pcie-xvc.c:411:30: error: 'PCI_EXT_CAP_ID_VNDR' undeclared (first use in this function); did you mean 'PCI_EXT_CAP_ID_LTR'?                                                                                            
   if (PCI_EXT_CAP_ID(cap) == PCI_EXT_CAP_ID_VNDR) {                                                                                             
                              ^~~~~~~~~~~~~~~~~~~                                                                                                
                              PCI_EXT_CAP_ID_LTR
```

JanMatCodasip, 2024-12-16:
The macro `PCI_EXT_CAP_ID_LTR` is defined in `linux/pci_regs.h`, and is included through `#include <linux/pci.h>`. The given commit builds correctly both on my RHEL8 machine and on the Github CI.

In conclusion, this looks to be an issue with your system or build configuration, not with riscv-openocd. I'm therefore closing this thread.

If you believe there is something wrong with riscv-openocd itself, please feel free to re-open this thread and describe the issue.

closed 2024-12-16 by JanMatCodasip

fanghuaqi, 2024-12-16:
Hi, I met the same issue, I am using Centos 6 to build this openocd, see 

~~~shell
[root@8c49e4bc022f ~]# uname -a
Linux 8c49e4bc022f 5.4.0-189-generic #209-Ubuntu SMP Fri Jun 7 14:05:13 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
# I cannot find PCI_EXT_CAP_ID_VNDR  macro, but I can find PCI_CAP_ID_VNDR    macro
[root@8c49e4bc022f ~]# grep -rn PCI_EXT_CAP_ID_VNDR /usr/include/linux/
[root@8c49e4bc022f ~]# grep -rn PCI_CAP_ID_VNDR /usr/include/linux/
/usr/include/linux/pci_regs.h:206:#define  PCI_CAP_ID_VNDR      0x09    /* Vendor specific */
~~~

I cannot find this macro in the `/usr/include/linux/pci_regs.h`, but I can find it in ubuntu 20.04's header file.

TommyMurphyTM1234, 2024-12-16:
CentOS 6 is c. 20 years old. 
I wouldn't use such an old distro.

TommyMurphyTM1234, 2024-12-16:
> I'm having problems trying to compile the latest OpenOCD here at [de20c2a](https://github.com/riscv-collab/riscv-openocd/commit/de20c2ad5faa0bc411b86bd6db69186b8c2ac979).
> 
> ```
> Work/openocd-build/openocd.git/src/jtag/drivers/xlnx-pcie-xvc.c: In function 'xlnx_pcie_xvc_init':                         
> Work/openocd-build/openocd.git/src/jtag/drivers/xlnx-pcie-xvc.c:411:30: error: 'PCI_EXT_CAP_ID_VNDR' undeclared (first use in this function); did you mean 'PCI_EXT_CAP_ID_LTR'?                                                                                            
>    if (PCI_EXT_CAP_ID(cap) == PCI_EXT_CAP_ID_VNDR) {                                                                                             
>                               ^~~~~~~~~~~~~~~~~~~                                                                                                
>                               PCI_EXT_CAP_ID_LTR
> ```

You'll probably need to provide more details about your host platform and maybe also the build log. But, maybe like the post above, you are simply using an outdated Linux distro that doesn't provide the necessary support?

JanMatCodasip, 2024-12-16:
Does the same issue occur when you try to build vanilla OpenOCD from openocd.org? If so, and if you believe that OpenOCD should support Centos6, please continue the discussion on OpenOCD mailing list.

However, Centos6 is end of life since November 2024, so I am afraid that OpenOCD community probably won't invest time into supporting this distro.

fanghuaqi, 2024-12-17:
Hi @TommyMurphyTM1234 @JanMatCodasip , I will investigate it later, and if could not fixed, I will upgrade my build environment, thanks