riscv-collab/riscv-openocd #1191: Single-core running “target_run_algorithm” in SMP architecture forcibly interrupted by ”riscv_openocd_poll“

Alan-19950616, 2024-12-25:
The run algorithm code segment.
https://github.com/riscv-collab/riscv-openocd/blob/riscv/src/target/riscv/riscv.c#L3482-L3521

The problematic code segment.
https://github.com/riscv-collab/riscv-openocd/blob/riscv/src/target/riscv/riscv.c#L3926-L3929

Alan-19950616, 2024-12-25:
This PR brought in
https://github.com/riscv-collab/riscv-openocd/pull/767

aap-sc, 2024-12-25:
@Alan-19950616 do you have a test to illustrate the issue?

Alan-19950616, 2024-12-25:
Of course, I can provide you with the logs.
[openocd.log](https://github.com/user-attachments/files/18244955/openocd.log)

aap-sc, 2024-12-25:
@Alan-19950616 logs of OpenOCD are not exactly the reproduction. Could you please provide the full reproduction? Configuration files used, commands passed to gdb, etc.

Alan-19950616, 2024-12-25:
Here is the configuration file, GDB only uses the load command.
[openocd_ns_cluster0_core0.cfg.txt](https://github.com/user-attachments/files/18245771/openocd_ns_cluster0_core0.cfg.txt)

TommyMurphyTM1234, 2024-12-25:
> Here is the configuration file, GDB only uses the load command

It can't be the only GDB command used as it surely has to at least use some commands to connect to the target in the first place? I think what @aap-sc is asking for is complete instructions and artifacts required for a reproducible test case. E.g.

- OpenOCD script (provided above) 
- the command line used to run OpenOCD 
- the command line used to run GDB
- the set of commands fed to GDB 
- the verbose `openocd -d3` log for the test case (provided above?)
- maybe details of your target RISC-V implementation?

en-sc, 2024-12-25:
@Alan-19950616, AFAIK this should fix the issue you are having:
https://github.com/tom-van/openocd-rp2350-riscv/commit/87bf74eecbb35efa57c9fa2342ce26114bf1d7a9

Please, try it out.

@tom-van, @Wren6991, please consider creating a PR to https://github.com/riscv-collab/riscv-openocd.

Alternatively, I can cherry-pick your commit and create a PR myself but it doesn't feel right :)

tom-van, 2024-12-25:
@en-sc thanks for interconnecting the topic!

> @tom-van, @Wren6991, please consider creating a PR

I'm not the author.
@Wren6991 promised to do so some time ago ;-)

Alan-19950616, 2024-12-26:
@en-sc Thanks, it does solve my problem. but verify failed.
```
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'gdb port', not 'gdb_port'
warning: multi-threaded target stopped without sending a thread-id, using first non-exited thread
__amp_wait () at ../../../SoC/ns_cluster0_core0/Common/Source/GCC/startup_ns_cluster0_core0.S:335
335         j 1b
(gdb) monitor flash write_image erase unlock D:\\openocd_test\\flash.bin 0x20000000 bin
auto erase enabled
auto unlock enabled
wrote 7077888 bytes from file D:\openocd_test\flash.bin in 160.736511s (43.002 KiB/s)
(gdb) monitor flash verify_image D:\\openocd_test\\flash.bin 0x20000000 bin
verify failed in bank at 0x20000000 starting at 0x00000000
Protocol error with Rcmd
(gdb)
```
I read the flash to a file via the “flash read_bank” command and then compared it to the original data passed.

en-sc, 2024-12-26:
@Alan-19950616,
1. Please, provide more information about your scenario (flash driver you are using, etc.).
2. Let's track this in a separate issue.

Alan-19950616, 2024-12-27:
Ok, I'll close the current issues and create another issue to upload the information.

en-sc, 2024-12-27:
I'd like to leave this issue open until [the fix](https://github.com/tom-van/openocd-rp2350-riscv/commit/87bf74eecbb35efa57c9fa2342ce26114bf1d7a9) is merged.

tom-van, 2025-05-19:
https://github.com/riscv-collab/riscv-openocd/pull/1259