riscv-collab/riscv-openocd #1195: Run "monitor flash verify_image" command failed in gdb

Alan-19950616, 2024-12-27:
1. The CPU is running in Flashxip mode and is halt at a hardware breakpoint position
2. Run `monitor flash write_image erase unlock D:\\openocd_test\\flash.bin 0x20000000 bin` in gdb
3. Run `monitor flash verify_image D:\\openocd_test\\flash.bin 0x20000000 bin` in gdb
4. At this point you can see the error message as follows

```
(gdb) load
Loading section .init, size 0x294 lma 0x20000000
Loading section .text, size 0x3720 lma 0x200002c0
Loading section .data, size 0xc0 lma 0x200039e0
Start address 0x0000000020000100, load size 14964
Transfer rate: 13 KB/sec, 4988 bytes/write.
(gdb) b main
Breakpoint 1 at 0x20001678: file main.c, line 99.
Note: automatically using hardware breakpoints for read-only addresses.
(gdb) c
Continuing.

Thread 1 "riscv0.cpu.0" hit Breakpoint 1, main () at main.c:99
99          hartid = __get_hart_id();
(gdb) monitor flash write_image erase unlock D:\\openocd_test\\flash.bin 0x20000000 bin
auto erase enabled
auto unlock enabled
wrote 7077888 bytes from file D:\openocd_test\flash.bin in 167.017609s (41.385 KiB/s)
(gdb) monitor flash verify_image D:\\openocd_test\\flash.bin 0x20000000 bin
verify failed in bank at 0x20000000 starting at 0x00000000
Protocol error with Rcmd
```

OpenOCD configuration file [openocd_ns_cluster0_core0.cfg.txt](https://github.com/user-attachments/files/18256661/openocd_ns_cluster0_core0.cfg.txt)
Error log file [openocd.log](https://github.com/user-attachments/files/18256664/openocd.log)
Flash Driver [src/flash/nor/nuspi.c](https://github.com/riscv-mcu/riscv-openocd/blob/nuclei/2024/src/flash/nor/nuspi.c)
Flash Loader [contrib/loaders/flash/nuspi](https://github.com/riscv-mcu/riscv-openocd/tree/nuclei/2024/contrib/loaders/flash/nuspi)

JanMatCodasip, 2025-01-29:
@Alan-19950616 

- What hardware can be used to reproduce the issue (e.g. a generally available dev board)?
- The openocd.log attached is not in a verbose mode. If you wish to get help with your issue, please collect a verbose log (argument `-d3`).

TommyMurphyTM1234, 2025-01-29:
```
Protocol error with Rcmd
```
Might this also be an issue with the GDB Remote Serial Protocol handling on the OpenOCD side? And, as such, the GDB Remote Serial Protocol verbose log might be useful? As well as the version of GDB being used?

Alan-19950616, 2025-02-10:
```
(gdb) load
Loading section .init, size 0x258 lma 0x20000000
Loading section .text, size 0x2b38 lma 0x20000280
Loading section .rodata, size 0x4a4 lma 0x20002db8
Loading section .data, size 0x6c lma 0x2000325c
Start address 0x200000c0, load size 12960
Transfer rate: 8 KB/sec, 3240 bytes/write.
(gdb) b main
Breakpoint 1 at 0x20001420: file main.c, line 94.
Note: automatically using hardware breakpoints for read-only addresses.
(gdb) c
Continuing.

Breakpoint 1, main () at main.c:94
94          srand(__get_rv_cycle()  | __get_rv_instret() | __RV_CSR_READ(CSR_MCYCLE));
(gdb) monitor flash write_image erase unlock D:\\openocd_test\\flash.bin 0x20000000 bin
auto erase enabled
auto unlock enabled
wrote 7077888 bytes from file D:\openocd_test\flash.bin in 334.709015s (20.651 KiB/s)
(gdb) monitor flash verify_image D:\\openocd_test\\flash.bin 0x20000000 bin
verify failed in bank at 0x20000000 starting at 0x00000000
(gdb) monitor flash read_bank 0 read.bin 0 0x6b2000
wrote 7020544 bytes to file read.bin from flash bank 0 at offset 0x00000000 in 270.081940s (25.385 KiB/s)
```

Write to flash bin file
[flash.bin.log](https://github.com/user-attachments/files/18732649/flash.bin.log)

Bin file read from flash
[read.bin.log](https://github.com/user-attachments/files/18732651/read.bin.log)

Partial log file zip for debug level 3
[openocd_d3.zip](https://github.com/user-attachments/files/18732709/openocd_d3.zip)

Alan-19950616, 2025-02-11:
The file is too big, I'm sharing it via cloud drive, it's valid for 1 day, click on the link to save it!

https://www.alipan.com/t/210FmnSXmYmmqpHjYgWK

Alan-19950616, 2025-02-11:
The file is too big, I'm sharing it via cloud drive, it's valid for 7 day, click on the link to save it!

https://www.alipan.com/s/c491PHMshLt

aap-sc, 2025-02-11:
> The file is too big, I'm sharing it via cloud drive, it's valid for 7 day, click on the link to save it!
> 
> https://www.alipan.com/s/c491PHMshLt

1. Could you please archive the file? Usually xz-based archivation produces reasonable results. tar -cJf <archive.tar.xz> should do the job.
2. The file should be available for all people for more than a week. This bug looks complex enough and it may take more than a week to investigate.
3. it seems that the site requires registration. This is not convenient. Please, consider sharing the file using some other hosting service (once it is properly compressed).
4. Have your tried to the size of the binary file that you try to upload? See my previous comment for the rationale.

aap-sc, 2025-02-11:
And another one. If the new file is attached, please add input file and memory dump from target (this is needed to identify what went from with batch execution).

Alan-19950616, 2025-02-11:
Bbinary-file and log-file after binary reduction
[read.zip](https://github.com/user-attachments/files/18746826/read.zip)
[write.zip](https://github.com/user-attachments/files/18746828/write.zip)
[openocd.zip](https://github.com/user-attachments/files/18746825/openocd.zip)