riscv-collab/riscv-openocd #1196: isspace/isxdigit breaks build under Cygwin

jlupien-draper, 2024-12-27:
The toolchain build went quite well under Cygwin, but I have hit a couple
of snags with openOCD.

The "isspace()" and "isxdigit()" macros have a compilation error under Cygwin,
a warning that a char index may not be used for arrays bombs the build for a number
of files.

This is caused by the implementation of these macros not taking this restriction into
account - most environments likely do not enforce it.

The fix for me was to cast the argument to these macros to uint16_t at the invocation.
This is likely not the best fix, putting the cast into the macro definition would be more
"permanent" but it was difficult to find the correct definition to modify, so I fixed the
invocations instead.

I have made a local branch with the fixes, I would be happy to push it up if desired.

aap-sc, 2024-12-27:
@jlupien-draper from what I can see `isspace` and `isxdigit` come from these files:

```
src/server/telnet_server.c
src/svf/svf.c
src/target/riscv/riscv.c
src/pld/efinix.c
src/pld/gatemate.c
src/target/image.c
```

These is target-independent part. riscv-openocd focuses on RISC-V support and generally do not make modifications to these files. If you think that your patches are useful - it's better to open MR in upstream OpenOCD here 
https://openocd.org/pages/repos.html

en-sc, 2024-12-27:
Hi! Thank you for the report.

However, AFAIU `isspace()` and `isxdigit()` are part of C standard library and are defined to be a function that takes an `int` (should be defined in '<ctype.h>'). Therefore this is an issue with the standard library implementation you are using.

Maybe this issue will be helpfull: https://github.com/mintty/mintty/issues/891

jlupien-draper, 2024-12-27:
> Hi! Thank you for the report.
> 
> However, AFAIU `isspace()` and `isxdigit()` are part of C standard library and are defined to be a function that takes an `int` (should be defined in '<ctype.h>'). Therefore this is an issue with the standard library implementation you are using.
> 
> Maybe this issue will be helpfull: [mintty/mintty#891](https://github.com/mintty/mintty/issues/891)

OK, that's quite helpful, I went looking for a bit to find the right macros to fix and did not see them in the repo. I will follow up soon - please keep the issue open for bit.

jlupien-draper, 2024-12-27:
This text in ctype.h gives the definite hint:
#ifndef __cplusplus
/* These macros are intentionally written in a manner that will trigger
   a gcc -Wall warning if the user mistakenly passes a 'char' instead
   of an int containing an 'unsigned char'.  Note that the sizeof will
   always be 1, which is what we want for mapping EOF to __CTYPE_PTR[0];
   the use of a raw index inside the sizeof triggers the gcc warning if
   __c was of type char, and sizeof masks side effects of the extra __c.
   Meanwhile, the real index to __CTYPE_PTR+1 must be cast to int,
   since isalpha(0x100000001LL) must equal isalpha(1), rather than being
   an out-of-bounds reference on a 64-bit machine.  */
#define __ctype_lookup(__c) ((__CTYPE_PTR+sizeof(""[__c]))[(int)(__c)])

So my fix was incorrect in any case, because the cast should be to int, not uint16_t,
and the issue is likely to be in the upstream openOCD repo as suggested by aap-sc above.
I will follow up there, thanks for the help.

closed 2024-12-27 by jlupien-draper