riscv-collab/riscv-openocd #1205: `riscv_read_by_any_size()` is redundant

en-sc, 2025-01-15:
`riscv_read_by_any_size()` is used to try all available sizes when reading memory:

https://github.com/riscv-collab/riscv-openocd/blob/eb9ba216e199a3a93f976246823c7a9653ad5064/src/target/riscv/riscv.c#L1392-L1397

`riscv_read_by_any_size()` is currently used in `riscv_semihosting()` and in `riscv_add_breakpoint()` to read instructions.

There is also `target_read_buffer()`. It provides the same interface.
Seems like `riscv_read_by_any_size()` should be the implementation of `target_read_buffer()` (currently `target_read_buffer_default()` is used).

JanMatCodasip, 2025-01-20:
Actually, the functions `riscv_read_by_any_size()` and `target_read_buffer_default()` work differently:

- `riscv_read_by_any_size()` tries to read certain amount of data from memory, and will try to use various access sizes until one of them succeeds. This function was introduced for architectures that do not support all access sizes when accessing special areas of memory that can hold program instructions (namely ICCM memory in [VeeR EH1](https://github.com/chipsalliance/Cores-VeeR-EH1)).
- `target_read_buffer_default()` will read certain amount of bytes using the largest transfer size possible in order to improve performance. It will **not** fall back to a different access size if the attempted access size is not supported - it will end with an error immediately.

en-sc, 2025-01-20:
> This function was introduced for architectures that do not support all access sizes when accessing special areas of memory that can hold program instructions

But the issue is `target_read_buffer/target_write_buffer` seems to be the functions intended to be used to write instruction memory.
This is evident from:
* The comment in `target.h`
https://github.com/riscv-collab/riscv-openocd/blob/eb9ba216e199a3a93f976246823c7a9653ad5064/src/target/target.h#L635-L641
* The use in `load_image` handler:
https://github.com/riscv-collab/riscv-openocd/blob/22108977c0714ea0d3a0c103028d3b2269a909b1/src/target/target.c#L3658-L3659

Moreover, they are used throughout `riscv.c` for exactly that purpose:
* In `riscv_checksum_memory()`:
https://github.com/riscv-collab/riscv-openocd/blob/22108977c0714ea0d3a0c103028d3b2269a909b1/src/target/riscv/riscv.c#L3804-L3805
*  In `riscv_hit_watchpoint()`:
https://github.com/riscv-collab/riscv-openocd/blob/22108977c0714ea0d3a0c103028d3b2269a909b1/src/target/riscv/riscv.c#L2367

Finally, I don't get why the function can't do both: try multiple access sizes to be able to read any memory and try the largest size first to achieve the best possible performance.

en-sc, 2025-01-20:
Somewhat related to #1184.