riscv-collab/riscv-openocd #1211: Enable access memory via a examined target before examed all smp targets

lz-bro, 2025-01-23:
```
set _TARGETNAME $_CHIPNAME.cpu
target create $_TARGETNAME.0 riscv -chain-position $_TARGETNAME -coreid 0 -rtos hwthread
target create $_TARGETNAME.1 riscv -chain-position $_TARGETNAME -coreid 1
target smp $_TARGETNAME.0 $_TARGETNAME.1

$_TARGETNAME.0 configure -event examine-end {
    halt
    // access memory
}
```

I would like to make the following modifications to support，or better method?
```
--- a/src/target/riscv/riscv.c
+++ b/src/target/riscv/riscv.c
@@ -2534,8 +2534,21 @@ int riscv_halt(struct target *target)

        LOG_TARGET_DEBUG(target, "halting all harts");

-       int result = ERROR_OK;
+       bool smp_targets_examined = false;
        if (target->smp) {
+               smp_targets_examined = true;
+               struct target_list *tlist;
+               foreach_smp_target(tlist, target->smp_targets) {
+                       struct target *t = tlist->target;
+                       if (!target_was_examined(t)) {
+                               smp_targets_examined = false;
+                               break;
+                       }
+               }
+       }
+
+       int result = ERROR_OK;
+       if (smp_targets_examined) {
                struct target_list *tlist;
                foreach_smp_target(tlist, target->smp_targets) {
                        struct target *t = tlist->target;

```

aap-sc, 2025-01-23:
you can just temporary disable smp groups (you have this smp on/off command , or you could create an smp group later).

lz-bro, 2025-02-05:
> you can just temporary disable smp groups (you have this smp on/off command , or you could create an smp group later).

The smp command should be before init, and how could I create an smp group later.

aap-sc, 2025-02-05:
> > you can just temporary disable smp groups (you have this smp on/off command , or you could create an smp group later).
> 
> The smp command should be before init, and how could I create an smp group later.

that's not true.

See the documentation here:

[image]

you can also take at discussion in this MR for additional information: https://github.com/riscv-collab/riscv-openocd/pull/872

lz-bro, 2025-02-05:
> > > you can just temporary disable smp groups (you have this smp on/off command , or you could create an smp group later).
> > 
> > 
> > The smp command should be before init, and how could I create an smp group later.
> 
> that's not true.
> 
> See the documentation here:
> 
> [image]
> 
> you can also take a look at discussion in this MR for additional information: [#872](https://github.com/riscv-collab/riscv-openocd/pull/872)

Thanks, it's helpful.