riscv-collab/riscv-openocd #1216: `deassert_reset` should not wait for a target to come out of reset and should not update `target->state`

en-sc, 2025-01-27:
Currently there is a loop in `deassert_reset()` handler that waits for the hart to come out of reset.
https://github.com/riscv-collab/riscv-openocd/blob/e870c5f2dece31d67bb1d69384cbb05d20a543ab/src/target/riscv/riscv-013.c#L2930-L2953

This is not necessary -- there is a call to `arp_waitstate`:
https://github.com/riscv-collab/riscv-openocd/blob/e870c5f2dece31d67bb1d69384cbb05d20a543ab/src/target/startup.tcl#L140

The task is to move handling of a target coming out of reset from `deassert_reset()` to `poll()`. This will also allow to drop the awkward handling of an unexpected reset in `riscv013_get_hart_state()`:
https://github.com/riscv-collab/riscv-openocd/blob/e870c5f2dece31d67bb1d69384cbb05d20a543ab/src/target/riscv/riscv-013.c#L2716-L2737

en-sc, 2025-01-27:
Huge thanks to @tom-van for pointing this out. Please see https://github.com/riscv-collab/riscv-openocd/pull/1214#issuecomment-2615828010 for the original comment.

tom-van, 2025-01-27:
Well at least Cortex-Ms `deassert_reset()` leaves TARGET_RESET state until `poll()` detects that CPU runs or is halted.
Unlike this Cortex-A calls poll() in `deassert_reset()` to update the state immediately.

RISC-V has much better designed req - ack structure of control/status and we are able to detect that reset is done.
Yes, I can imagine `deassert_reset()` just clears NMDRESET and the rest of processing is done in the following  poll().

Anyway the reset assert/deassert implementation in OpenOCD has serious shortcomings for multi-cores and needs significant rewrite. I tried once, but don't have enough power to uphold it.