riscv-collab/riscv-openocd #1223: Problem in halting multiple RISC-V harts

jimaandro, 2025-02-06:
Hello.
I have an issue when I try to run OpenOCD to multicore RISC-V. My target is an Openpiton System with 4 CVA6 cores.
I am using `Open On-Chip Debugger 0.12.0`
I built the OpenOCD using the default configuration and adding the extra flag `--enable-xlnx-pcie-xvc and --prefix=$HOME`

I used the configuration file below

```
adapter speed 1000

adapter driver ftdi
ftdi vid_pid 0x0403 0x6010
ftdi channel 0
transport select jtag
ftdi tdo_sample_edge falling
ftdi layout_init 0x0008 0x000b
reset_config none

set _CHIPNAME riscv
jtag newtap $_CHIPNAME cpu -irlen 5

set _TARGETNAME $_CHIPNAME.cpu

set _TARGETNAME_0 $_CHIPNAME.cpu0
set _TARGETNAME_1 $_CHIPNAME.cpu1
set _TARGETNAME_2 $_CHIPNAME.cpu2
set _TARGETNAME_3 $_CHIPNAME.cpu3

# Create targets for each hart
target create $_TARGETNAME_0 riscv -chain-position $_TARGETNAME -coreid 0 -rtos hwthread
target create $_TARGETNAME_1 riscv -chain-position $_TARGETNAME -coreid 1
target create $_TARGETNAME_2 riscv -chain-position $_TARGETNAME -coreid 2
target create $_TARGETNAME_3 riscv -chain-position $_TARGETNAME -coreid 3

# Configure work area for debugging
$_TARGETNAME_0 configure -work-area-phys 0x80000000 -work-area-size 1000 -work-area-backup 1
$_TARGETNAME_1 configure -work-area-phys 0x80001000 -work-area-size 1000 -work-area-backup 1
$_TARGETNAME_2 configure -work-area-phys 0x80002000 -work-area-size 1000 -work-area-backup 1
$_TARGETNAME_3 configure -work-area-phys 0x80003000 -work-area-size 1000 -work-area-backup 1

target smp $_TARGETNAME_0 $_TARGETNAME_1 $_TARGETNAME_2 $_TARGETNAME_3

gdb_report_data_abort enable
gdb_report_register_access_error enable

riscv set_reset_timeout_sec 120
riscv set_command_timeout_sec 120

# Try enabling address translation (only works for newer versions)
if { [catch {riscv set_enable_virtual on} ] } {
    echo "Warning: This version of OpenOCD does not support address translation. To debug on virtual addresses, please update to the latest version." }

init
halt
echo "Ready for Remote Connections"

```

My problem is that I get
`**Core x could not be made part of halt group 1.**` as you can see in log file I attached

[openOCD.log](https://github.com/user-attachments/files/18694593/openOCD.log)

Using this configuration, **_only the last hart/target that is created is halted (in this case it is hart 3)._**
As you can see, only hart 3 is set to ZSBL address (0xfff1010000), and only the registers of this hart are 0.

[image]

cpu0 registers:

[image]

cpu3 registers:

[image]

I want all 4 harts to stay halted in ZSBL address with all registers equal to 0. The reason is that I am trying to boot Linux using OpenOCD. I have managed some times to boot, but this happens only if all harts stay on ZSBL during openOCD halt command. I think the problem here is the 
`**Core x could not be made part of halt group 1.**`

How can I fix this?

TommyMurphyTM1234, 2025-02-06:
> `**Core x could not be made part of halt group 1.**` as you can see in log file I attached
> 
> [openOCD.log](https://github.com/user-attachments/files/18694593/openOCD.log)
```
Open On-Chip Debugger 0.12.0+dev-01578-g2c8376b79 (2024-05-17-15:10)
```
You should probably build and try the latest OpenOCD.

jimaandro, 2025-02-06:
I still get the same results

[image]

aap-sc, 2025-02-06:
@jimaandro  I haven't looked at the logs yet. Me or someone else will certainly do it once they have time (@en-sc , @JanMatCodasip FYI)

In the meantime I have few additional questions and suggestions.

1. "**Core x could not be made part of halt group 1.** this is not necessarily an error. Most likely this means that your system does not support a concept of "halt groups" (this should be clarified one we analyse the logs and/or consult the documentation for your system). This is an optional feature and it is fine not to support it. I suggest to double-check the documentation for your system (if you can) if it should be supported. 

Halt groups is a feature that allows debugger to halt or resume several harts (almost) simultaneously.  If this feature is not supported and you try to enable smp - OpenOCD will still try to halt/resume all harts, but it will do it one by one - leading to a significant delay between the respected operation.

2. what is ZSBL? is it some kind of a proprietary bootloader?

3. if I understand your logs correctly, then all your harts are actually halted. It's just that addresses are different. You can double-check this by issuing the following command in your gdb session:

```
monitor targets
```

It should display target state from OpenOCD point of view.

4. If you want all you harts to be on the same address. you could try to perform `reset halt` command right after init.
```
init
reset halt
```

depending on your design and you configuration this may allow you to have more control on which hart goes where.

TommyMurphyTM1234, 2025-02-07:
> 2\. what is ZSBL? is it some kind of a proprietary bootloader?

I presume that it's Zero Stage Boot Loader - the earliest stage of a bootloader that may be split into several logical/physical stages? E.g.:

- https://github.com/riscv-software-src/opensbi/blob/master/docs/external/coreboot.md

I'm not sure that what's actually executing on the target matters in the context of this issue? Unless what the ZSBL does somehow impacts OpenOCD debugging behaviour?

aap-sc, 2025-02-07:
```
I'm not sure that what's actually executing on the target matters in the context of this issue? Unless what the ZSBL does somehow impacts OpenOCD debugging behaviour?
```

From my point of view:

1. it looks like (**I still can  be wrong**) user is incorrect in his assessment that only one hart is halt. From the description it looks like all harts are actually in a halted state
2. To me it looks like that something goes wrong with the boot process - and this leads to other harts being at unexpected places. This behavior may depend on the bootloader .

I hope this clarifies the rationale.

jimaandro, 2025-02-07:
Hello @aap-sc 

Ok I understand what you are saying here
> 1. "**Core x could not be made part of halt group 1.** this is not necessarily an error. Most likely this means that your system does not support a concept of "halt groups" (this should be clarified [...]

I will try to see if Halt groups are available. If it's not supported, is there any way to enable it?
The way OpenOCD halts harts looks like it does not halt all harts at the same time

> 2. what is ZSBL? is it some kind of a proprietary bootloader?
Yes. As @TommyMurphyTM1234 mentioned, ZSBL is Zero Stage Boot Loader. It's the bootROM of the RISC-V target. The ZSBL does not affect the OpenOCD, the problem is that I am trying to boot Linux using OpenOCD, and to achieve this all cores must be halted when they are on the ZSBL stage. 
I don't understand why only one hart is halted on the ZSBL address (0xfff1010000), while the other 3 continue the execution till they are halted on a different address.

> 3. if I understand your logs correctly, then all your harts are actually halted. It's just that addresses are different. You can double-check this by issuing the following command in your gdb session:> 

Yes all harts are halted, but on different addresses. And because of this, the harts that where halted after the ZSBL (harts 0,1,2), have registers (like ra, sp) different from 0. So when I type `continue` (after Linux image is loaded), the booting stacks in the stage of bringing up secondary CPUs. That's because these cores have not actually been reset, since their registers are not 0x0.

I have managed to fully boot linux, only when randomly all the harts stay halted at 0xfff1010000 (ZSBL). But this is completely random.

> 4. If you want all you harts to be on the same address. you could try to perform `reset halt` command right after init.
> 
> ```
> init
> reset halt
> ```

Yes This is actually what I want. I want all harts to be reset on same initial address 0xfff1010000 (ZSBL).
I tried the `reset halt` but still I get the same results. Only the last target I create is halted at 0xfff1010000

aap-sc, 2025-02-07:
```
I will try to see if Halt groups are available. If it's not supported, is there any way to enable it?
```

If these are not available, there is no way to enable it (unless this functionality require some custom initialization/unlock routine).

```
The way OpenOCD halts harts looks like it does not halt all harts at the same time
```

Yes, this is the expected behavior if halt groups are not available.

```
I tried the reset halt but still I get the same results. Only the last target I create is halted at 0xfff1010000
```

Got it. What is the state of your design right after `reset halt` ? Could you modify your initialization script as follows:

```
init
reset halt
for { set i 0 } { $i < 4 } { incr i } {  targets $i ;   echo "pc after reset halt: [reg pc]" }
```

jimaandro, 2025-02-07:
> Got it. What is the state of your design right after `reset halt` ? Could you modify your initialization script as follows:
> 
> ```
> init
> reset halt
> for { set i 0 } { $i < 4 } { incr i } {  targets $i ;   echo "pc after reset halt: [reg pc]" }
> ```

After `reset halt` I get these PC values:

```
Open On-Chip Debugger 0.12.0+dev-01578-g2c8376b79-dirty (2025-02-06-23:38)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : Hardware thread awareness created
Info : clock speed 1000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x249511c3 (mfg: 0x0e1 (Wintec Industries), part: 0x4951, ver: 0x2)
Info : datacount=2 progbufsize=8
Info : Core 0 could not be made part of halt group 1.
Info : Examined RISC-V core; found 4 harts
Info :  hart 0: XLEN=64, misa=0x800000000014112d
Info : [riscv.cpu0] Examination succeed
Info : datacount=2 progbufsize=8
Info : Core 1 could not be made part of halt group 1.
Info : Examined RISC-V core; found 4 harts
Info :  hart 1: XLEN=64, misa=0x800000000014112d
Info : [riscv.cpu1] Examination succeed
Info : datacount=2 progbufsize=8
Info : Core 2 could not be made part of halt group 1.
Info : Examined RISC-V core; found 4 harts
Info :  hart 2: XLEN=64, misa=0x800000000014112d
Info : [riscv.cpu2] Examination succeed
Info : datacount=2 progbufsize=8
Info : Core 3 could not be made part of halt group 1.
Info : Examined RISC-V core; found 4 harts
Info :  hart 3: XLEN=64, misa=0x800000000014112d
Info : [riscv.cpu3] Examination succeed
Info : starting gdb server for riscv.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: riscv.cpu tap/device found: 0x249511c3 (mfg: 0x0e1 (Wintec Industries), part: 0x4951, ver: 0x2)
pc after reset halt: pc (/64): 0x00000000800048ea
pc after reset halt: pc (/64): 0x00000000800048ea
pc after reset halt: pc (/64): 0x00000000800048ea
pc after reset halt: pc (/64): 0x000000fff1010000
Ready for Remote Connections
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
```

If I change the order of target creation
For example if I change the target 3 to be created after target 0

```
# Create targets for each hart
target create $_TARGETNAME_0 riscv -chain-position $_TARGETNAME -coreid 0 -rtos hwthread
target create $_TARGETNAME_3 riscv -chain-position $_TARGETNAME -coreid 3
target create $_TARGETNAME_1 riscv -chain-position $_TARGETNAME -coreid 1
target create $_TARGETNAME_2 riscv -chain-position $_TARGETNAME -coreid 2
```

Then the reset halt returns different order of values
```
pc after reset halt: pc (/64): 0x00000000800048ea
pc after reset halt: pc (/64): 0x00000000800048ea
pc after reset halt: pc (/64): 0x000000fff1010000
pc after reset halt: pc (/64): 0x00000000800048ea
```

As you can see, only the last target that was created is halted properly

aap-sc, 2025-02-07:
```
As you can see, only the last target that was created is halted properly
```

1. Ok. I'm not if this can be classified as "not halted properly". For now we can only say that value of program counters upon reset are not what is expected. The root cause is still unknown, ultimately we need to double check the documentation for your system or maybe double-check reset configuration.

2. Do you mind to try yet another suggestion? You can try to set program counters manually:

```
init
reset halt
for { set i 0 } { $i < 4 } { incr i } {  targets $i ;   reg pc 0xfff1010000 }
```

What happens in this case? Can you boot the system?

jimaandro, 2025-02-07:
> 2. Do you mind to try yet another suggestion? You can try to set program counters manually:
> 
> ```
> init
> reset halt
> for { set i 0 } { $i < 4 } { incr i } {  targets $i ;   reg pc 0xfff1010000 }
> ```
> 
> What happens in this case? Can you boot the system?

Yes this works!
Thank you very much!

I had tried to set the PC to 0xfff1010000 via GDB, but I got the busy error, and GDB then got stuck in a loop. But in OpenOCD it works fine.