riscv-collab/riscv-openocd #1237: Why `riscv-011_reg.c` don't have `examine_vlenb` and `examine_mtopi` API？

Alan-19950616, 2025-03-06:
```
Debug: 109 464 riscv.c:2295 riscv_examine(): [riscv.cpu] Starting examination
Debug: 110 464 riscv.c:441 dtmcontrol_scan(): [riscv.cpu] DTMCS: 0x0 -> 0x1450
Debug: 111 464 riscv.c:2309 riscv_examine(): [riscv.cpu] dtmcontrol=0x1450
Debug: 112 464 riscv.c:2311 riscv_examine(): [riscv.cpu] version=0x0
Debug: 113 464 riscv-011.c:2390 init_target(): init
Debug: 114 464 riscv_reg.c:709 riscv_reg_impl_init_cache(): [riscv.cpu] create register cache for 4194 registers
Debug: 115 483 server.c:608 sig_handler(): Terminating on Signal 22
Assertion failed!
Expression: false && "Existence of other registers is determined " "depending on existence of these ones, so " "whether these register exist or not should be " "set explicitly."
```

https://github.com/riscv-collab/riscv-openocd/blob/riscv/src/target/riscv/riscv_reg.c#L368-L379

`examine_vlenb`
https://github.com/riscv-collab/riscv-openocd/blob/riscv/src/target/riscv/riscv-013_reg.c#L126-L163

 `examine_mtopi`
https://github.com/riscv-collab/riscv-openocd/blob/riscv/src/target/riscv/riscv-013_reg.c#L267-L290

aap-sc, 2025-03-06:
@Alan-19950616 why do you ask? Discovery of these vector registers were never support for 0.11 targets. Recent changes just made this explicit to avoid crashes on non RISCV-V compliant targets.

Do you have a 0.11-compliant HW that supports RVV? If this is the case - then file a bug, please.

en-sc, 2025-03-06:
@Alan-19950616, AFAIU this is the issue fixed in https://github.com/riscv-collab/riscv-openocd/pull/1207, please, try using a more recent OpenOCD version.

aap-sc, 2025-03-06:
> [@Alan-19950616](https://github.com/Alan-19950616), AFAIU this is the issue fixed in [#1207](https://github.com/riscv-collab/riscv-openocd/pull/1207), please, try using a more recent OpenOCD version.

@en-sc , @Alan-19950616 the issue fixed by #1207 addresses this assertion failure:

```
Expression: false && "Existence of other registers is determined " "depending on existence of these ones, so " "whether these register exist or not should be " "set explicitly."
```

However, vlenb/mtopei/mtopi are discovery mechanism is still not available for riscv-0.11 targets. Just FYI

Alan-19950616, 2025-03-06:
Yes, my riscv-0.11 target also doesn't support the vlenb/mtopei/mtopi registers, using the latest openocd version solves the problem of the connection being ASSERTED!

closed 2025-03-06 by Alan-19950616