riscv-collab/riscv-openocd #1242: GDB+OpenCD remote debug cause access a section of address starting from 0

lz-bro, 2025-03-27:
I use GDB+OpenCD for remote debugging of a board, which runs Linux system. When GDB makes a `si` or `register write`, etc. which will cause access a section of address starting from 0, and OpenOCD print a bunch of errors about invalid PTE. I'm just confused at this point, and if anyone can help me, I'd be really grateful.

```
GNU gdb (GDB) 14.2
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=riscv64-unknown-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) target remote 10.0.90.107:1024
Remote debugging using 10.0.90.107:1024
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0xffffffff80fdce50 in ?? ()
(gdb) set debug remote 1
(gdb) si
[remote] Sending packet: $m0,2#fb
[remote] Packet received: 0000
[remote] Sending packet: $m2,2#fd
[remote] Packet received: 0000
[remote] Sending packet: $m4,2#ff
[remote] Packet received: 0000
[remote] Sending packet: $m6,2#01
[remote] Packet received: 0000
[remote] Sending packet: $m8,2#03
[remote] Packet received: 0000
[remote] Sending packet: $ma,2#2c
[remote] Packet received: 0000
[remote] Sending packet: $mc,2#2e
[remote] Packet received: 0000
[remote] Sending packet: $me,2#30
[remote] Packet received: 0000
[remote] Sending packet: $m10,2#2c
[remote] Packet received: 0000
[remote] Sending packet: $m12,2#2e
[remote] Packet received: 0000
[remote] Sending packet: $m14,2#30
[remote] Packet received: 0000
[remote] Sending packet: $m16,2#32
[remote] Packet received: 0000
[remote] Sending packet: $m18,2#34
[remote] Packet received: 0000
[remote] Sending packet: $m1a,2#5d
[remote] Packet received: 0000
[remote] Sending packet: $m1c,2#5f
[remote] Packet received: 0000
[remote] Sending packet: $m1e,2#61
[remote] Packet received: 0000
[remote] Sending packet: $m20,2#2d
[remote] Packet received: 0000
[remote] Sending packet: $m22,2#2f
[remote] Packet received: 0000
[remote] Sending packet: $m24,2#31
[remote] Packet received: 0000
[remote] Sending packet: $m26,2#33
[remote] Packet received: 0000
[remote] Sending packet: $m28,2#35
[remote] Packet received: 0000
[remote] Sending packet: $m2a,2#5e
[remote] Packet received: 0000
[remote] Sending packet: $m2c,2#60
[remote] Packet received: 0000
[... 257 lines cut ...]
[remote] Packet received: 0000
[remote] Sending packet: $m54,2#34
[remote] Packet received: 0000
[remote] Sending packet: $m56,2#36
[remote] Packet received: 0000
[remote] Sending packet: $m58,2#38
[remote] Packet received: 0000
[remote] Sending packet: $m5a,2#61
[remote] Packet received: 0000
[remote] Sending packet: $m5c,2#63
[remote] Packet received: 0000
[remote] Sending packet: $m5e,2#65
[remote] Packet received: 0000
[remote] Sending packet: $m60,2#31
[remote] Packet received: 0000
[remote] Sending packet: $m62,2#33
[remote] Packet received: 0000
[remote] Sending packet: $mffffffff80fdc67e,2#62
[remote] Packet received: 0111
[remote] Sending packet: $mffffffff80fdc67c,2#60
[remote] Packet received: edb7
[remote] Sending packet: $mffffffff80fdc67a,2#5e
[remote] Packet received: 0290
0xffffffff80fdc67e in ?? ()
```

TommyMurphyTM1234, 2025-03-27:
That rings a bell for me but at the moment I can't find anything on it. I have a feeling that this happens early on (at examine time?) for some reason. Does it happen again if you do a second or subsequent `si`?

Can you capture, zip (if it's large) and upload the `openocd -d3` verbose log for such an interaction to see exactly what's going on?

You're doing `si` without having loaded any program to the target. Did you intend to do that? 

> ```
> warning: No executable has been specified and target does not support determining executable automatically.  Try using the "file" command.
> ```

What are the register values after connecting but before issuing the `si` command?

zhangjing0303, 2025-03-27:
I have encountered the same problem. This usually happens when gdb is just connected through the target remote command, and gdb does not have a symbol table yet.

lz-bro, 2025-03-27:
> That rings a bell for me but at the moment I can't find anything on it. I have a feeling that this (reading of 100 (?) bytes starting at address `0x00000000`) happens early on (at examine time?) for some reason. Does it happen again if you do a second or subsequent `si`?
> 
> Can you capture, zip (if it's large) and upload the `openocd -d3` verbose log for such an interaction to see exactly what's going on?
> 
> You're doing `si` without having loaded any program to the target. Did you intend to do that?
> 
> > ```
> > warning: No executable has been specified and target does not support determining executable automatically.  Try using the "file" command.
> > ```
> 
> What are the register values after connecting but before issuing the `si` command?

1. It will happen again when I do the second step.
2. openocd log has been uploaded
3. This situation occurred when I did not load any symbol table.

This seems to be an issue with GDB，some information from gdb source(gdb/riscv-tdep.c)

```
/* The prologue scanner.  This is currently only used for skipping the
   prologue of a function when the DWARF information is not sufficient.
   However, it is written with filling of the frame cache in mind, which
   is why different groups of stack setup instructions are split apart
   during the core of the inner loop.  In the future, the intention is to
   extend this function to fully support building up a frame cache that
   can unwind register values when there is no DWARF information.  */

static CORE_ADDR
riscv_scan_prologue (struct gdbarch *gdbarch,
                     CORE_ADDR start_pc, CORE_ADDR end_pc,
                     struct riscv_unwind_cache *cache)
{
  CORE_ADDR cur_pc, next_pc, after_prologue_pc;
  CORE_ADDR original_end_pc = end_pc;
  CORE_ADDR end_prologue_addr = 0;

  /* Find an upper limit on the function prologue using the debug
     information.  If the debug information could not be used to provide
     that bound, then use an arbitrary large number as the upper bound.  */
  after_prologue_pc = skip_prologue_using_sal (gdbarch, start_pc);
  if (after_prologue_pc == 0)
    after_prologue_pc = start_pc + 100;   /* Arbitrary large number.  */
  if (after_prologue_pc < end_pc)
    end_pc = after_prologue_pc;

```