riscv-collab/riscv-openocd #1249: Thread 1 "openocd" received signal SIGSEGV, Segmentation fault.

Alan-19950616, 2025-04-30:
Setting debug_level greater than or equal to 3 before init in the openocd configuration file results in a “Segmentation fault” error.

1. Error Output
```
Debug: 176 9 batch.c:270 log_batch(): 41b nop 0x00000000 @nop -> success 0x03ff0001 @dmcontrol; 4i
Debug: 177 9 batch.c:190 log_dmi_decoded(): read: dmcontrol=0x3ff0001 {dmactive=active hartsello=0x3ff}
Debug: 178 9 batch.c:304 riscv_batch_run_from(): [riscv.cpu] Running batch of scans [0, 2)
Debug: 179 9 batch.c:270 log_batch(): 41b write 0x00000001 @dmcontrol -> success 0x00000000 @nop; 4i
Debug: 180 9 batch.c:190 log_dmi_decoded(): write: dmcontrol=1 {dmactive=active}
Debug: 181 9 batch.c:270 log_batch(): 41b nop 0x00000000 @nop -> success 0x00000001 @dmcontrol; 4i
Debug: 182 9 batch.c:304 riscv_batch_run_from(): [riscv.cpu] Running batch of scans [0, 2)
Debug: 183 10 batch.c:270 log_batch(): 41b read 0x00000000 @dmstatus -> success 0x00000000 @nop; 4i
Debug: 184 10 batch.c:270 log_batch(): 41b nop 0x00000000 @nop -> success 0x004003a2 @dmstatus; 4i
Debug: 185 10 batch.c:190 log_dmi_decoded(): read: dmstatus=0x4003a2 {version=0_13 impebreak=1 hasresethaltreq=1 authenticated=true anyhalted=1 allhalted=1}
Debug: 186 10 batch.c:304 riscv_batch_run_from(): [riscv.cpu] Running batch of scans [0, 2)

Thread 1 "openocd" received signal SIGSEGV, Segmentation fault.
__strlen_avx2 () at ../sysdeps/x86_64/multiarch/strlen-avx2.S:96
96      ../sysdeps/x86_64/multiarch/strlen-avx2.S: No such file or directory.
(gdb) bt
#0  __strlen_avx2 () at ../sysdeps/x86_64/multiarch/strlen-avx2.S:96
#1  0x00007ffff7a6cd15 in __vfprintf_internal (s=s@entry=0x7fffffffbcf0, format=format@entry=0x71e680 "%db %s 0x%08x @%s -> %s 0x%08x @%s; %di", ap=ap@entry=0x7fffffffbe78,
    mode_flags=mode_flags@entry=0) at vfprintf-internal.c:1688
#2  0x00007ffff7a7ff9a in __vsnprintf_internal (string=0x7fffffffbde0 "41b write 0x00000020 @", maxlen=<optimized out>, format=0x71e680 "%db %s 0x%08x @%s -> %s 0x%08x @%s; %di",
    args=0x7fffffffbe78, mode_flags=0) at vsnprintf.c:114
#3  0x0000000000497bc3 in ?? ()
#4  0x0000000000497cfe in ?? ()
#5  0x0000000000497df7 in ?? ()
#6  0x00000000005db8f3 in ?? ()
#7  0x0000000000596c38 in ?? ()
#8  0x0000000000596dff in ?? ()
#9  0x000000000058e91c in ?? ()
#10 0x000000000059a752 in ?? ()
#11 0x000000000047827d in ?? ()
#12 0x000000000046574f in ?? ()
#13 0x000000000046589e in ?? ()
#14 0x0000000000407a15 in ?? ()
#15 0x0000000000495ad7 in ?? ()
#16 0x00000000005f8fb9 in ?? ()
#17 0x0000000000000001 in ?? ()
```

2. OpenOCD Base Commit
`https://github.com/riscv-collab/riscv-openocd/commit/f82c5a7`

3. OS Message
```
NAME="Ubuntu"
VERSION="20.04.6 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.6 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal
```

TommyMurphyTM1234, 2025-04-30:
> 2\. OpenOCD Base Commit
>     `https://github.com/riscv-collab/riscv-openocd/commit/f82c5a7`

Why such an old commit?
Does the problem persist if you build from the latest top of tree?

If it does then maybe build OpenOCD with debug enabled in case it provides more information in the stack trace?
And/or add some custom logging to see more details of the data preceding the crash?

Alan-19950616, 2025-05-07:
Compiling with the latest code from the riscv branch, the “Segmentation fault” error has been fixed!

closed 2025-05-07 by Alan-19950616