riscv-collab/riscv-openocd #1250: jimtcl Building error

jimaandro, 2025-05-03:
Hello
I am trying to build riscv-opeocd, but something is going wrong during bootstrap.
I ran these:
```
git clone https://github.com/riscv-collab/riscv-openocd.git
cd riscv-openocd/
./bootstrap
./configure
```

`bootstrap `returns:
```
+ aclocal --warnings=all
+ libtoolize --automake --copy
+ autoconf --warnings=all
+ autoheader --warnings=all
+ automake --warnings=all --gnu --add-missing --copy
configure.ac:24: installing 'build-aux/compile'
configure.ac:42: installing 'build-aux/config.guess'
configure.ac:42: installing 'build-aux/config.sub'
configure.ac:19: installing 'build-aux/install-sh'
configure.ac:19: installing 'build-aux/missing'
Makefile.am: installing './INSTALL'
Makefile.am: installing 'build-aux/depcomp'
Makefile.am:26: installing 'build-aux/mdate-sh'
Makefile.am:26: installing 'build-aux/texinfo.tex'
Skipping submodule setup
Bootstrap complete. Quick build instructions:
./configure ....
```

and `configure ` returns:
```
checking for makeinfo... makeinfo
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of gcc... gcc3
checking for ranlib... ranlib
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.23... yes
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-pc-linux-gnu file names to x86_64-pc-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for file... file
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... (cached) ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /usr/bin/dd
checking how to truncate binary pipes... /usr/bin/dd bs=4096 count=1
checking for mt... mt
checking if mt is a manifest tool... no
checking for stdio.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for strings.h... yes
checking for sys/stat.h... yes
checking for sys/types.h... yes
checking for unistd.h... yes
checking for dlfcn.h... yes
checking for objdir... .libs
checking if gcc supports -fno-rtti -fno-exceptions... no
checking for gcc option to produce PIC... -fPIC -DPIC
checking if gcc PIC flag -fPIC -DPIC works... yes
checking if gcc static flag -static works... yes
checking if gcc supports -c -o file.o... yes
checking if gcc supports -c -o file.o... (cached) yes
checking whether the gcc linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking dynamic linker characteristics... GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking whether stripping libraries is possible... yes
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... no
checking whether to build static libraries... yes
checking for an ANSI C-conforming const... yes
checking for unsigned long long int... yes
checking for long long int... yes
checking for library containing ioperm... none required
checking for library containing dlopen... none required
checking for library containing openpty... none required
checking for sys/socket.h... yes
checking for elf.h... yes
checking for Elf64_Ehdr... yes
checking for glibc... yes
checking for fcntl.h... yes
checking for malloc.h... yes
checking for netdb.h... yes
checking for poll.h... yes
checking for strings.h... (cached) yes
checking for sys/ioctl.h... yes
checking for sys/param.h... yes
checking for sys/select.h... yes
checking for sys/stat.h... (cached) yes
checking for sys/sysctl.h... no
checking for sys/time.h... yes
checking for sys/types.h... (cached) yes
checking for unistd.h... (cached) yes
checking for arpa/inet.h... yes
checking for netinet/in.h... yes
checking for netinet/tcp.h... yes
checking whether to enable assertions... yes
checking for _Bool... yes
checking for stdbool.h that conforms to C99... yes
checking whether byte ordering is bigendian... no
checking for strndup... yes
checking for strnlen... yes
checking for gettimeofday... yes
checking for usleep... yes
checking for realpath... yes
checking whether to build a release... no
checking whether to build Doxygen as HTML... yes
checking whether to build Doxygen as PDF... no
checking whether to enable verbose USB I/O messages... no
checking whether to enable verbose USB communication messages... no
checking whether to enable malloc free space logging... no
checking for jimtcl >= 0.79... no
checking for jim.h... no
configure: error: jimtcl is required but not found via pkg-config and system includes

```

I also tried `git submodule update --init --recursive` before `bootstrap`
and also I tried manually running `./configure` and `make` inside jimtcl folder before running `./configure` in main folder.
Nothing of these worked. Am I doing something wrong?

aap-sc, 2025-05-03:
@jimaandro see this commit: https://github.com/riscv-collab/riscv-openocd/commit/77c904fd13c08077cafd0845107506db408b5bb1

Basically an internal copy of jim-tcl (from the submodule is deprecated now). You can still use it, though, if you pass `--enable-internal-jimtcl` to  your `./configure` invocation. Though I would advise to adjust your workflow and avoid doing that - since there are plans to remove jim-tcl from the submodue.

jimaandro, 2025-05-03:
Ok @aap-sc thank you very much!

closed 2025-05-03 by jimaandro