riscv-collab/riscv-openocd #1251: Getting a "Error: RTOS: failed to get register X" message

Hershkyman, 2025-05-04:
Hi,
Running in SMP mode, after I load my program, and try to read using the memory Browser, I get the following Error messages:

[image]
And the list goes on.

Looking at riscv.c, the function riscv_init_registers is responsible to set if a CSR is valid, enabled.
So for these specific CRS(s) there are no special cases, so they remain as the default: exist=true.

I wanted to know if anyone has encountered this issue and is working on setting these CRS(s) properly.
Here is a partial list of these CRS(s) with relevant names:

"
case CSR_SEED:
case CSR_SEDELEG:
case CSR_SIDELEG:
/* virtual supervisor group crs(s) */
case CSR_VSSTATUS:
case CSR_VSIE:
case CSR_VSTVEC:
case CSR_VSSCRATCH:
/* hyper-extension group crs(s) */
case CSR_MSTATUSH:
"

TommyMurphyTM1234, 2025-05-04:
What is "the memory Browser"?

If, as I suspect, it's part of some GUI IDE (Eclipse?), then can you reproduce the problem when running OpenOCD standalone outside of such an environment?

You should probably capture, zip and upload a verbose `openocd -d3` log for analysis.

en-sc, 2025-05-05:
@Hershkyman,
```
/ hyper-extension group crs(s) */
case CSR_MSTATUSH:
```
This does not seem right. AFAIK, `mstatush` is just the high bits of `mstatus` on 32-bit target.
Am I missing something? If not, please adjust the description.

en-sc, 2025-05-05:
@Hershkyman, thank you for reporting this issue!
.
As you have mentioned, some registers are mistakenly assumed to exist and the existence of these register can be derived relatively painlessly:
E.g.:
* `mstatush` exists only on 32-bit targets.
* Virtual supervisor CSRs existence is indicated by `misa.H` bit.

These issues can be addressed. However, for some registers (e.g. `seed`), the only method to know whether it exists or not is to try to access it.

Anyway, as a workaround, `riscv hide_csrs` command can be used to mark problematic registers as hidden. If you are already using `riscv hide_csrs`, please share a `-d3` log as suggested by @TommyMurphyTM1234.