riscv-collab/riscv-openocd #1275: Memory Sampling support for targets with last level write-back cache(s)

RaamyPi, 2025-07-08:
Greetings.

The memory sample command in openocd polls a memory location for updates, but it seems like the current implementation has overlooked the aspect of memory hierarchy in targets. In our experiments, the memory sample command constantly read stale values from memory when in fact the memory location was being written to (verified from commit log), reason being - we have a last level write-back cache that doesn't evict lines unless replaced/flushed. 
 
Between samples, openocd could have sent fences by doing silent debug mode switches (via progbuf) but that does not seem to be happening (dmstatus values in between sampling always show harts running and never halting). These fences would have flushed the dirty lines from the cache to memory and stale values would not have been read. 
 
Can we have appropriate fixes for memory sampling support for targets with last level write-back cachesas well? 
 
Thanks.

RaamyPi, 2025-07-13:
I am not sure if the problem is clear. 

Let me try to explain it better with an example.

`i = 0;`
`while (1) i++;`

Say I want to sample the variable i. I find out what address this variable is mapped to and I use the memory sample command to sample this location.

What's expected to happen? The memory sample command should show samples that are incrementing.

What happens on a system without last level writeback caches? The memory sample command shows me samples that are incrementing.

What happens on a system with last level writeback caches? The memory sample command will only show me 0, because once a block gets accomodated into the cache, updates never reach the lower level (memory here) unless that block is evicted.

Why does this happen? This is a direct consequence of the features of sysbus that you mentioned, sysbus gives us "sideband" access to memory, it's lightweight and fast. But I don't think it's right to be used for memory sampling (in the way it is now) because it is misleading. The command is supposed to tell me how a memory location is changing but it is not and it will only under certain conditions (no writeback).

Anyways, thank you for your suggestion.

I leave it up the collaborators to decide how to address this.

aap-sc, 2025-07-15:
> What happens on a system without last level writeback caches? The memory sample command shows me samples that are incrementing.

@RaamyPi formally, fences won't help you here. Fences guarantee only **ordering**, **provided that some other effect was observed** that was in the program after the barrier. They do not guarantee that writeback will happen.  

What are asking is related to cache management. Historically   RISC-V specification avoided cache maintenance operations - it was up to implementation. Yes, recently there was Zicbom but it is quite a recent addition and is optional.

It don't think it is possible to implement what your are asking in a portable manner.