riscv-collab/riscv-openocd #1278: Enable riscv virt2phys_mode to hw, insert breakpoing failed.

weiqi7777, 2025-08-07:
hi:
we use the hardware translate virtual address to access memory.
as below:
[image]

when we execute si command in gdb, it print the warning, and gdb will not work after the warning. 

We reproduced this scenario on the hardware and found that OpenOCD made the hardware modify the instruction region by walking the page table and inserting ebreak instructions. However, this instruction region's page table attributes lack the Write (W) permission, causing the write operation to fail.

So how should we solve the warning?

thanks

weiqi7777, 2025-08-08:
If the hw virt2phys is not enable,  openocd will use read the page to translate the virtual address to physical address, then use physicall address to write ebreak instruction into instruction region memory, that will not cause page fault exception. 

But if hw virt2phys is enable, opencod will let hardware to translate the address, then to write ebreak instruction into instruction region memory，that will cause page fault exception.

As you mentioned, use the hardware breakpoint, that will limit the breakpoint number use,  because hardware breakpoint is limit. 

So we are curious, does OpenOCD not support this mode?

aap-sc, 2025-08-08:
@weiqi7777 

I think this is a defect actually, I'm not quite sure that it can be fixed easily, though.

@fk-sc, @en-sc , @MarekVCodasip    FYI.

In my opinion for this specific case the final behavior should be _functionally equivalent_ to the case when a SW translation mode was selected. Using HW breakpoints is not usable in all use cases due to resource constraints.

The case of **setting**/**removing** breakpoints alone can be fixed quite easily. GDB by itself does not give direct commands to read/write memory when setting SW breakpoints.  Instead, GDB sends Z0/z0 packets to add/remove sw breakpoints. It is up to OpenOCD to decide what to do on behalf of these commands.

However, there is a more serious issue with GDB/OpenOCD in this mode. Before sending Z0/z0 packets GDB tries to read memory in order to detect "atomic region" (lr/sc sequence) to skip it. This is done using "memory read" commands (via `x` packet of gdb remote serial protocol). Since this is a separate packet - I don't see how we can differentiate it from ordinary read/writes. This specific issue may require more substantial changes in the codebase than a simple fix.

So for now I suggest to keep the issue open. I'm going to take a look at this issue in a greater detail once I have some time. For now I would treat this mode as not compatible with GDB.

aap-sc, 2025-08-08:
> However, for instruction region accesses or modifications (such as inserting ebreak instructions), software should be used to perform the address translation to obtain the physical address corresponding to the PC, and then use this physical address to access or modify the memory.

Yes, the problem is that in **GDB remote protocol** there is no distinction between "instruction" and "data" addresses. The protocol uses just "memory access". Currently in OpenOCD there is no facilities to even describe the current memory map.  That's the crux of the issue.